mchehab/zbar #226: Allow Sbinary and --xml together

petarmarj, 2022-08-27:
Thank you for your valuable work on the zbarimg project. My problem is the following:

**Version** 0.23.91
**OS**: MacOS 12.5.1

I need to read QR codes that are only scanned with `-Sbinary` in the correct character set encoding. 

Without this function, the word `Zürich` in a Swiss-QR-Bill (see maybe [this issue](https://github.com/mchehab/zbar/issues/216)) will be read as `Z羹rich`. My limited skills in PHP make it impossible for me to figure out how to correct this miscoding into the correct UTF-8 `Zürich`. Therefore, I am deeply grateful for the introduction of this parameter.

However, the `-Sbinary` option prevents the `--xml` output. This format would be desirable in my project because it allows me to read multiple QR codes in one document. The `--raw` output enforced by `-Sbinary` merges all QR code contents, so I don't know where which content starts.

Is there already a workaround for this situation? I have already tried to adapt the C code - in [zbarimg.c#L244](https://github.com/mchehab/zbar/blob/master/zbarimg/zbarimg.c#L244) there is the if condition, which could probably be commented out. 

https://github.com/mchehab/zbar/blob/9f0cbc70aba627afe02cb0afdb8fbc39eb4f229d/zbarimg/zbarimg.c#L244-L251

But I fail at compilation (`clang: error: unsupported option '-fopenmp'`). Moreover, I suspect that other people might be happy about this function - hence this issue. Thank you again for your work and greetings from Z羹rich.

matheusmoreira, 2023-09-28:
> The `--raw` output enforced by `-Sbinary` merges all QR code contents, so I don't know where which content starts.

Yes. With text data, you can have separator characters such as the `\n` line feeds present in the code you referenced. Binary data is different though: any byte is valid, even nulls.

So you have to decode each QR code separately and capture the individual outputs. That way you know where each one begins and ends. Incidentally, this is exactly why I wrote one shot mode to accompany binary decoding mode.

The *true* problem is the text in your QR code is being decoded incorrectly for some reason. The binary decoding option fixes it because it bypasses the character encoding conversion logic.

tormodvolden, 2024-08-23:
> The _true_ problem is the text in your QR code is being decoded incorrectly for some reason.

See also #281. The root cause is that text streams that are both valid latin1 and big5 are decoded as big5 (or sjis).

JamesR98, 2024-10-05:
This may be better off in a new issue but it seems to be related to 'find out where multiple qr codes start and end':

> Yes. With text data, you can have separator characters such as the `\n` line feeds present in the code you referenced. Binary data is different though: any byte is valid, even nulls.

I'm not sure if there's a reason file output hasn't been added to zbarimg, but it strikes me as a solution:

`zbarimg -Sbinary -q --raw multiple.png -o data.bin
`
and output:

```
data.bin_1
data.bin_2
data.bin_3
```

matheusmoreira, 2024-10-05:
@JamesR98

Simplicity, I suppose. To contribute the oneshot and binary decoding modes, I had to modify an old, complex codebase I was unfamiliar with. I went straight for the simplest solution I could think of:

 - Bypass the charset conversion logic when decoding binary QR codes
 - Write the contents of exactly one QR code to standard output

This set of functionality allows writing shell scripts that run the zbar tools in a loop and saves the output to a file until it exits with a non-successful error code. That way I didn't have to write it in C. Avoided a lot of work. And also allows the user to control the file names and how sequencing is done.

Building this into zbar itself could be useful. It could serve as the foundation of a structured append implementation.