mchehab/zbar #234: Error Installing Zbar in python

sc0rp10n-py, 2022-11-02:
I want to install zbar in python and use it in python.

I did run

`python setup.py install` but i get this error

```
C:\Program Files\Python310\python.exe: can't open file 'C:\\Users\\princ\\Downloads\\projects\\freelancing\\fiverr\\zbar\\setup.py': [Errno 2] No such file or directory
PS C:\Users\princ\Downloads\projects\freelancing\fiverr\zbar> cd .\zbar\
PS C:\Users\princ\Downloads\projects\freelancing\fiverr\zbar\zbar> cd .\pygtk\
PS C:\Users\princ\Downloads\projects\freelancing\fiverr\zbar\zbar\pygtk> cd ..
PS C:\Users\princ\Downloads\projects\freelancing\fiverr\zbar\zbar> cd .\python\
PS C:\Users\princ\Downloads\projects\freelancing\fiverr\zbar\zbar\python> python setup.py install
C:\Users\princ\Downloads\projects\freelancing\fiverr\zbar\zbar\python\setup.py:2: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.core import setup, Extension
running install
running build
running build_ext
building 'zbar' extension
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I../include -IC:\Program Files\Python310\include -IC:\Program Files\Python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcdecoder.c /Fobuild\temp.win-amd64-3.10\Release\decoder.obj
decoder.c
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I../include -IC:\Program Files\Python310\include -IC:\Program Files\Python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcenum.c /Fobuild\temp.win-amd64-3.10\Release\enum.obj
enum.c
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I../include -IC:\Program Files\Python310\include -IC:\Program Files\Python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcexception.c /Fobuild\temp.win-amd64-3.10\Release\exception.obj
exception.c
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I../include -IC:\Program Files\Python310\include -IC:\Program Files\Python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcimage.c /Fobuild\temp.win-amd64-3.10\Release\image.obj
image.c
image.c(335): warning C4244: 'function': conversion from 'Py_ssize_t' to 'unsigned long', possible loss of data
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I../include -IC:\Program Files\Python310\include -IC:\Program Files\Python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcimagescanner.c /Fobuild\temp.win-amd64-3.10\Release\imagescanner.obj
imagescanner.c
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I../include -IC:\Program Files\Python310\include -IC:\Program Files\Python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcprocessor.c /Fobuild\temp.win-amd64-3.10\Release\processor.obj
processor.c
processor.c(244): warning C4244: '=': conversion from 'double' to 'long', possible loss of data
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I../include -IC:\Program Files\Python310\include -IC:\Program Files\Python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcscanner.c /Fobuild\temp.win-amd64-3.10\Release\scanner.obj
scanner.c
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I../include -IC:\Program Files\Python310\include -IC:\Program Files\Python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcsymbol.c /Fobuild\temp.win-amd64-3.10\Release\symbol.obj
symbol.c
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I../include -IC:\Program Files\Python310\include -IC:\Program Files\Python310\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcsymboliter.c /Fobuild\temp. [... cut ...] ld\temp.win-amd64-3.10\Release\symboliter.obj build\temp.win-amd64-3.10\Release\symbolset.obj build\temp.win-amd64-3.10\Release\zbarmodule.obj /OUT:build\lib.win-amd64-3.10\zbar.cp310-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.10\Release\zbar.cp310-win_amd64.lib
LINK : fatal error LNK1181: cannot open input file 'zbar.lib'
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed with exit code 1181
```

sc0rp10n-py, 2022-11-02:
zbar.lib is missing in windows kits path
where can i find that?

sc0rp10n-py, 2022-11-04:
Hey @mchehab, can you please help with this.

sgn, 2022-11-04:
I'm not doing Windows. In Linux, we build python binding as part of zbar. The `zbar.lib`, I suppose it's equivalent with Linux `libzbar.a`, which is built during that process.

sc0rp10n-py, 2022-11-04:
@sgn  So it gets built during running this command `python setup.py install` ?

sgn, 2022-11-04:
No, the upper level, there's a readme for windows there https://github.com/mchehab/zbar/blob/master/README-windows.md

sc0rp10n-py, 2022-11-04:
@sgn 
[image]

How do I suppose run this in windows? `pacman` is not available for windows.
```
pacman -Syu --noconfirm autoconf libtool automake make \
autoconf-archive pkg-config autopoint gettext-devel
```

sgn, 2022-11-04:
I don't do Windows, sorry.

sc0rp10n-py, 2023-01-18:
> I have built one python helper for zbar.You can follow my helper https://github.com/wqh17101/zbar_lite you can install by `pip install zbar-lite` for some basic ability. @sc0rp10n-py

Hey
what all features are not their as per your knowledge?

wqh17101, 2023-01-19:
@sc0rp10n-py Nope. all the features in zbar-lite is based on zbar.

sc0rp10n-py, 2023-01-19:
> @sc0rp10n-py Nope. all the features in zbar-lite is based on zbar.

No I am asking, what all features does zbar have extra when compared to zbar-lite?

wqh17101, 2023-01-20:
Some video-related features.
Zbar-lite only provides the  image-related features , because video features can be implemented by `opencv and zbar's image-related feature in python` @sc0rp10n-py