mchehab/zbar #235: Error building in windows

sc0rp10n-py, 2022-11-04:
After chocolatey install, msys2 wasn't opening so i downloaded from the website
then in that i opened msys2 mingw64

There I installed the dependencies from pacman

Then ran this
`autoreconf -vfi`

Now i ran this
```
./configure --host=i686-w64-mingw32 --prefix=/usr/local/win32 --with-gtk --with-python --with-qt --without-java --without-imagemagick --enable-pthread  --with-directshow --disable-dependency-tracking
```

Now the below error comes.

```
princ@lappy MINGW64 /c/Users/princ/Downloads/projects/freelancing/fiverr/zbar1/zbar
$ make
make  all-recursive
make[1]: Entering directory '/c/Users/princ/Downloads/projects/freelancing/fiverr/zbar1/zbar'
Making all in zbar
make[2]: Entering directory '/c/Users/princ/Downloads/projects/freelancing/fiverr/zbar1/zbar/zbar'
  CC       processor/libzbar_la-win.lo
In file included from ./processor.h:26,
                 from processor/win.c:29:
../include/config.h:423: warning: "_WIN32_WINNT" redefined
  423 | #define _WIN32_WINNT 0x0500
      |
In file included from /usr/include/w32api/windows.h:9,
                 from processor/win.c:28:
/usr/include/w32api/_mingw.h:239: note: this is the location of the previous definition
  239 | #define _WIN32_WINNT 0x601
      |
In file included from /usr/include/w32api/winnt.h:9,
                 from /usr/include/w32api/minwindef.h:163,
                 from /usr/include/w32api/windef.h:9,
                 from /usr/include/w32api/windows.h:69,
                 from processor/win.c:28:
processor/win.c: In function ‘_zbar_event_init’:
processor/win.c:41:14: error: incompatible types when assigning to type ‘zbar_event_t’ {aka ‘struct zbar_event_s’} from type ‘HANDLE’ {aka ‘void *’}
   41 |     *event = CreateEvent(NULL, 0, 0, NULL);
      |              ^~~~~~~~~~~
processor/win.c:42:22: error: used struct type value where scalar is required
   42 |     return ((*event) ? 0 : -1);
      |                      ^
processor/win.c: In function ‘_zbar_event_destroy’:
processor/win.c:47:9: error: used struct type value where scalar is required
   47 |     if (*event) {
      |         ^
processor/win.c:48:21: error: incompatible type for argument 1 of ‘CloseHandle’
   48 |         CloseHandle(*event);
      |                     ^~~~~~
      |                     |
      |                     zbar_event_t {aka struct zbar_event_s}
In file included from /usr/include/w32api/winbase.h:19,
                 from /usr/include/w32api/windows.h:70,
                 from processor/win.c:28:
/usr/include/w32api/handleapi.h:19:49: note: expected ‘HANDLE’ {aka ‘void *’} but argument is of type ‘zbar_event_t’ {aka ‘struct zbar_event_s’}
   19 |   WINBASEAPI WINBOOL WINAPI CloseHandle (HANDLE hObject);
      |                                          ~~~~~~~^~~~~~~
In file included from /usr/include/time.h:16,
                 from /usr/include/pthread.h:14,
                 from ./mutex.h:104,
                 from ./event.h:27,
                 from ./processor.h:36,
                 from processor/win.c:29:
processor/win.c:49:18: error: incompatible types when assigning to type ‘zbar_event_t’ {aka ‘struct zbar_event_s’} from type ‘void *’
   49 |         *event = NULL;
      |                  ^~~~
processor/win.c: In function ‘_zbar_event_trigger’:
processor/win.c:55:14: error: incompatible type for argument 1 of ‘SetEvent’
   55 |     SetEvent(*event);
      |              ^~~~~~
      |              |
      |              zbar_event_t {aka struct zbar_event_s}
In file included from /usr/include/w32api/winbase.h:35,
                 from /usr/include/w32api/windows.h:70,
                 from processor/win.c:28:
/usr/include/w32api/synchapi.h:47:46: note: expected ‘HANDLE’ {aka ‘void *’} but argument is of type ‘zbar_event_t’ {aka ‘struct zbar_event_s’}
   47 |   WINBASEAPI WINBOOL WINAPI SetEvent (HANDLE hEvent);
      |                                       ~~~~~~~^~~~~~
processor/win.c: In function ‘_zbar_event_wait’:
processor/win.c:65:30: error: incompatible type for argument 1 of ‘WaitForSingleObject’
   65 |     rc = WaitForSingleObject(*event, _zbar_timer_check(timeout));
      |                              ^~~~~~
      |                              |
      |                              zbar_event_t {aka struct zbar_event_s}
In file included from /usr/include/w32api/winbase.h:35,
                 from /usr/include/w32api/windows.h:70,
                 from processor/win.c:28:
/usr/include/w32api/synchapi.h:115:55: note: expected ‘HANDLE’ {aka ‘void *’} but argument is of type ‘zbar_event_t’ {aka ‘struct zbar_event_s’}
  115 |   WINBASEAPI DWORD WINAPI WaitForSingleObject (HANDLE hHandle, DWORD dwMilliseconds);
      |                                                ~~~~~~~^~~~~~~
processor/win.c: In function ‘_zbar_thread_start’:
processor/win.c:87:34: warning: passing argument 3 of ‘CreateThread’ from incompatible pointer type [-Wincompatible-pointer-types]
   87 |     hthr = CreateThread(NULL, 0, proc, arg, 0, NULL);
      |                                  ^~~~
      |                                  |
      |                                  void * (*)(void *)
In file included from /usr/include/w32api/winbase.h:29,
                 from /usr/include/w32api/windows.h:70,
                 from processor/win.c:28:
/usr/include/w32api/processthreadsapi.h:279:127: note: expected ‘LPTHREAD_START_ROUTINE’ {aka ‘unsigned int (*)(void *)’} but argument is of type ‘void * (*)(void *)’
  279 |   WINBASEAPI HANDLE WINAPI CreateThread (LPSECURITY_ATTRIBUTES lpThreadAttributes, SIZE_T dwStackSize, LPTHREAD_START_ROUTINE lpStartAddress, LPVOID lpParameter, DWORD dwCreationFlags, LPDWORD lpThreadId);
      |                                                                                                        ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from ./processor.h:35,
                 from processor/win.c:29:
processor/win.c: In function ‘win_handle_event’:
./error.h:112:29: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 4 has type ‘LONG’ {aka ‘int’} [-Wformat=]
  112 |             fprintf(stderr, "%s: " format, __func__, ##args); \
      |                             ^~~~~~
./error.h:112:29: note: in definition of macro ‘zprintf’
  112 |             fprintf(stderr, "%s: " format, __func__, ##args); \
      |                             ^~~~~~
./error.h:112:29: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 5 has type ‘LONG’ {aka ‘int’} [-Wformat=]
  112 |             fprintf(stderr, "%s: " format, __func__, ##args); \
      |                             ^~~~~~
./error.h:112:29: note: in definition of macro ‘zprintf’
  112 |             fprintf(stderr, "%s: " format, __func__, ##args); \
      |                             ^~~~~~
processor/win.c: In function ‘_zbar_processor_input_wait’:
processor/win.c:225:43: warning: passing argument 2 of ‘MsgWaitForMultipleObjects’ from incompatible pointer type [-Wincompatible-pointer-types]
  225 |     int rc = MsgWaitForMultipleObjects(n, event, 0, timeout, QS_ALLINPUT);
      |                                           ^~~~~
      |                                           |
      |                                           zbar_event_t * {aka struct zbar_event_s *}
In file included from /usr/include/w32api/windows.h:72,
                 from processor/win.c:28:
/usr/include/w32api/winuser.h:2988:80: note: expected ‘void * const*’ but argument is of type ‘zbar_event_t *’ {aka ‘struct zbar_event_s *’}
 2988 |   WINUSERAPI DWORD WINAPI MsgWaitForMultipleObjects(DWORD nCount,CONST HANDLE *pHandles,WINBOOL fWaitAll,DWORD dwMilliseconds,DWORD dwWakeMask);
      |                                                                                ^
processor/win.c: In function ‘_zbar_event_init’:
processor/win.c:43:1: warning: control reaches end of non-void function [-Wreturn-type]
   43 | }
      | ^
make[2]: *** [Makefile:1140: processor/libzbar_la-win.lo] Error 1
make[2]: Leaving directory '/c/Users/princ/Downloads/projects/freelancing/fiverr/zbar1/zbar/zbar'
make[1]: *** [Makefile:1900: all-recursive] Error 1
make[1]: Leaving directory '/c/Users/princ/Downloads/projects/freelancing/fiverr/zbar1/zbar'
make: *** [Makefile:980: all] Error 2

```

kamakazikamikaze, 2022-12-09:
I encountered the same issue with Cygwin. I'm not a C/C++ guy so it took a while, and this may be wrong, but it seems that `_WIN32` is not defined by the compiler.

The following line hinted towards this:

```

processor/win.c:49:18: error: incompatible types when assigning to type ‘zbar_event_t’ {aka ‘struct zbar_event_s’} from type ‘void *’
   49 |         *event = NULL;

```

`zbar_event_s` is defined for *nix hosts, not Windows. You can define this macro yourself when executing `make`. Simply use `CFLAGS="-D _WIN32" make`. This results in a successful compile for me (albeit with a number of warnings). Still working with the PyZbar project to verify that everything works as expected.

sc0rp10n-py, 2022-12-13:
@kamakazikamikaze 
Hey I got some ahead with this, but still this error is coming
when i run

`CFLAGS="-D _WIN32" make`

```
  CC       processor/libzbar_la-win.lo
In file included from ./processor.h:26,
                 from processor/win.c:29:
../include/config.h:423: warning: "_WIN32_WINNT" redefined
  423 | #define _WIN32_WINNT 0x0500
      |
In file included from /usr/include/w32api/windows.h:9,
                 from processor/win.c:28:
/usr/include/w32api/_mingw.h:239: note: this is the location of the previous definition
  239 | #define _WIN32_WINNT 0x601
      |
In file included from /usr/include/w32api/winnt.h:9,
                 from /usr/include/w32api/minwindef.h:163,
                 from /usr/include/w32api/windef.h:9,
                 from /usr/include/w32api/windows.h:69,
                 from processor/win.c:28:
processor/win.c: In function ‘_zbar_event_init’:
processor/win.c:41:14: error: incompatible types when assigning to type ‘zbar_event_t’ {aka ‘struct zbar_event_s’} from type ‘HANDLE’ {aka ‘void *’}
   41 |     *event = CreateEvent(NULL, 0, 0, NULL);
      |              ^~~~~~~~~~~
processor/win.c:42:22: error: used struct type value where scalar is required
   42 |     return ((*event) ? 0 : -1);
      |                      ^
processor/win.c: In function ‘_zbar_event_destroy’:
processor/win.c:47:9: error: used struct type value where scalar is required
   47 |     if (*event) {
      |         ^
processor/win.c:48:21: error: incompatible type for argument 1 of ‘CloseHandle’
   48 |         CloseHandle(*event);
      |                     ^~~~~~
      |                     |
      |                     zbar_event_t {aka struct zbar_event_s}
In file included from /usr/include/w32api/winbase.h:19,
                 from /usr/include/w32api/windows.h:70,
                 from processor/win.c:28:
/usr/include/w32api/handleapi.h:19:49: note: expected ‘HANDLE’ {aka ‘void *’} but argument is of type ‘zbar_event_t’ {aka ‘struct zbar_event_s’}
   19 |   WINBASEAPI WINBOOL WINAPI CloseHandle (HANDLE hObject);
      |                                          ~~~~~~~^~~~~~~
In file included from /usr/include/time.h:16,
                 from /usr/include/pthread.h:14,
                 from ./mutex.h:104,
                 from ./event.h:27,
                 from ./processor.h:36,
                 from processor/win.c:29:
processor/win.c:49:18: error: incompatible types when assigning to type ‘zbar_event_t’ {aka ‘struct zbar_event_s’} from type ‘void *’
   49 |         *event = NULL;
      |                  ^~~~
processor/win.c: In function ‘_zbar_event_trigger’:
processor/win.c:55:14: error: incompatible type for argument 1 of ‘SetEvent’
   55 |     SetEvent(*event);
      |              ^~~~~~
      |              |
      |              zbar_event_t {aka struct zbar_event_s}
In file included from /usr/include/w32api/winbase.h:35,
                 from /usr/include/w32api/windows.h:70,
                 from processor/win.c:28:
/usr/include/w32api/synchapi.h:47:46: note: expected ‘HANDLE’ {aka ‘void *’} but argument is of type ‘zbar_event_t’ {aka ‘struct zbar_event_s’}
   47 |   WINBASEAPI WINBOOL WINAPI SetEvent (HANDLE hEvent);
      |                                       ~~~~~~~^~~~~~
processor/win.c: In function ‘_zbar_event_wait’:
processor/win.c:65:30: error: incompatible type for argument 1 of ‘WaitForSingleObject’
   65 |     rc = WaitForSingleObject(*event, _zbar_timer_check(timeout));
      |                              ^~~~~~
[... 32 lines cut ...]
./error.h:112:29: note: in definition of macro ‘zprintf’
  112 |             fprintf(stderr, "%s: " format, __func__, ##args); \
      |                             ^~~~~~
processor/win.c: In function ‘_zbar_processor_input_wait’:
processor/win.c:225:43: warning: passing argument 2 of ‘MsgWaitForMultipleObjects’ from incompatible pointer type [-Wincompatible-pointer-types]
  225 |     int rc = MsgWaitForMultipleObjects(n, event, 0, timeout, QS_ALLINPUT);
      |                                           ^~~~~
      |                                           |
      |                                           zbar_event_t * {aka struct zbar_event_s *}
In file included from /usr/include/w32api/windows.h:72,
                 from processor/win.c:28:
/usr/include/w32api/winuser.h:2988:80: note: expected ‘void * const*’ but argument is of type ‘zbar_event_t *’ {aka ‘struct zbar_event_s *’}
 2988 |   WINUSERAPI DWORD WINAPI MsgWaitForMultipleObjects(DWORD nCount,CONST HANDLE *pHandles,WINBOOL fWaitAll,DWORD dwMilliseconds,DWORD dwWakeMask);
      |                                                                                ^
processor/win.c: In function ‘_zbar_event_init’:
processor/win.c:43:1: warning: control reaches end of non-void function [-Wreturn-type]
   43 | }
      | ^
make[2]: *** [Makefile:1140: processor/libzbar_la-win.lo] Error 1
make[2]: Leaving directory '/c/Users/princ/Downloads/projects/freelancing/fiverr/zbar/zbar'
make[1]: *** [Makefile:1900: all-recursive] Error 1
make[1]: Leaving directory '/c/Users/princ/Downloads/projects/freelancing/fiverr/zbar'
make: *** [Makefile:980: all] Error 2

```

sc0rp10n-py, 2022-12-15:
@kamakazikamikaze do you know about this issue?

kamakazikamikaze, 2022-12-15:
Maybe you need to prepend it to your configure command: `CFLAGS="-D _WIN32" ./configure ...`. Then call `CFLAGS="-D _WIN32" make` again

sc0rp10n-py, 2022-12-16:
@kamakazikamikaze can you tell your steps exactly what u did? can be simple English wording steps, its fine if you don't remember the commands exactly.

sc0rp10n-py, 2022-12-16:
i did prepend that to make

i got this error now

https://pastebin.com/dTtuY8Av

wqh17101, 2023-01-17:
I have built one python helper for zbar.You can follow my helper https://github.com/wqh17101/zbar_lite
you can install by pip install zbar-lite for some basic ability. @sc0rp10n-py