mchehab/zbar #239: zbarcam crashes on ubuntu server

bedy93, 2022-12-27:
Hi all,

I was trying to use zbarcam on some notebooks but it crashes.
OS: ubuntu 22.04 lts server & 22.10 server  Zbarcam: 0.23.92

I'm using the following command: zbarcam --nodisplay --verbose=3
It gives the following log: https://pastebin.com/pcZcJHUG  After v4l2_get_supported_resolutions fails in v4l2_probe_iomode, which leads to a segmentation fault.

I thought I dig into the issue, so I've installed the tools built my own version.
Surprisingly that leads to a different failiure(Probably because I haven't set HAVE_SYS_MMAN_H switch...?), see: https://pastebin.com/Nrd71Ebt 
The v4l2_probe_iomode finishes, but doesn't set vdo->iomode so after the probe_iomode finished it defaults to READWRITE, but it still fails.

I've hardcoded the idomode to MMAP and with that it runs just fine. See: https://pastebin.com/0USHcj0h
It also works with USERPTR. (No log saved) Strange that it defaults to READWRITE, which is the only not working option.

So my conclusion is, that there are two issues:
- The v4l2_probe_iomode fails during probing.
- The fallback idomode in v4l2_probe_iomode failure case is READWRITE which is not working (at least on my config)

Could you fix this? I can offer some support.

jsfan3, 2023-08-18:
I confirm the same bug on Linux Mint 21.

Harvie, 2023-11-08:
I have same issue on `Debian 12` (package from debian repo), zbarcam version `0.23.92`. It only happens with `--nodisplay` parameter. Causes segmentation fault when trying to acquire pthread mutex:

```
(gdb) bt full
#0  0x0000fffff7e31e64 in ___pthread_mutex_lock (mutex=0x98) at ./nptl/pthread_mutex_lock.c:80
        type = <optimized out>
        __PRETTY_FUNCTION__ = "___pthread_mutex_lock"
        id = <optimized out>
#1  0x0000fffff7f6aa58 in zbar_negotiate_format () from /lib/aarch64-linux-gnu/libzbar.so.0
No symbol table info available.
#2  0x0000fffff7f6c894 in zbar_processor_init () from /lib/aarch64-linux-gnu/libzbar.so.0
No symbol table info available.
#3  0x0000aaaaaaaa1ba4 in main (argc=<optimized out>, argv=<optimized out>) at zbarcam/zbarcam.c:307
        video_device = 0xaaaaaaaa2a18 ""
        dbus = 1
        display = 0
        infmt = 0
        outfmt = 0
        i = 4
        active = <optimized out>
```

Harvie, 2023-11-08:
Probably related to calling `window_lock()` when in `--nodisplay` mode?

Harvie, 2023-11-08:
Which was probably fixed in 8c9b148c15276d9e99ca1f5f8915e52b25b46765
@bedy93 @jsfan3  can you test please?

chairwa, 2024-12-13:
Fixde by version 0.23.93 zbar-tools on ubuntu 24.04.