mchehab/zbar #263: CVE-2023-40890 & CVE-2023-40889

wqh17101, 2023-08-30:
there may be some vulns.
https://nvd.nist.gov/vuln/detail/CVE-2023-40889
https://nvd.nist.gov/vuln/detail/CVE-2023-40890
Please confirm and fix it if necessary.

sandrinr, 2023-10-19:
There is a fork addressing these two CVEs in a crude but hopefully effective manner: https://github.com/Raemi/zbar/commits/master

risicle, 2023-11-17:
The patch for CVE-2023-40890 in that branch is overly strict and causes zbar's tests to fail.

bastien-roucaries, 2023-11-29:
@risicle Patch for CVE-2023-40890 is not acceptable for a library: it print to stderr and exit

bastien-roucaries, 2023-11-29:
@risicle may be this one but I do not know how to test:
```
commit f613d7453cd1c2dd913bcc2181991a934eacbb60
Author: Remi Meier <remi.meier@xorlab.com>
Date:   Thu Oct 19 22:35:13 2023 +0000

    Add bounds check for CVE-2023-40890
    
    Add a check to avoid exploitation of the CVE.
    
    [debian] do not hardcode size of array and return -1 in case of error
    
    bug-debian-security: https://security-tracker.debian.org/tracker/CVE-2023-40889
    origin: https://github.com/mchehab/zbar/commit/d5857d3f7d0f5a243c517ee22762e1e3ddeb8db2

diff --git a/zbar/decoder/databar.c b/zbar/decoder/databar.c
index 8cfa848..2990e3b 100644
--- a/zbar/decoder/databar.c
+++ b/zbar/decoder/databar.c
@@ -23,6 +23,8 @@
 
 #include <config.h>
 #include <zbar.h>
+#include <stdlib.h>
+#include <stdio.h>
 
 #ifdef DEBUG_DATABAR
 # define DEBUG_LEVEL (DEBUG_DATABAR)
@@ -663,7 +665,7 @@ match_segment (zbar_decoder_t *dcode,
     return(ZBAR_DATABAR);
 }
 
-static inline unsigned
+static inline signed
 lookup_sequence (databar_segment_t *seg,
                  int fixed,
                  int seq[22])
@@ -676,6 +678,13 @@ lookup_sequence (databar_segment_t *seg,
     dbprintf(2, " {%d,%d:", i, n);
     p = exp_sequences + i;
 
+    if (n >= sizeof(seq)/sizeof(seq[0])-1) {
+	// The loop below checks i<n and increments i by one within the loop
+	// when accessing seq[22]. For this to be safe, n needs to be < 21.
+	// See CVE-2023-40890.
+	return -1;
+    }
+
     fixed >>= 1;
     seq[0] = 0;
     seq[1] = 1;
@@ -755,10 +764,15 @@ match_segment_exp (zbar_decoder_t *dcode,
             }
 
             if(!i) {
-                if(!lookup_sequence(seg, fixed, seq)) {
+                signed int lu = lookup_sequence(seg, fixed, seq);
+                if(!lu) {
                     dbprintf(2, "[nf]");
                     continue;
                 }
+                if(lu == -1) {
+                    dbprintf(1, " [aborted]\n");
+                    goto abort;
+                }
                 width = seg->width;
                 dbprintf(2, " A00@%d", j);
             }
@@ -829,6 +843,8 @@ match_segment_exp (zbar_decoder_t *dcode,
     dcode->direction = (1 - 2 * (seg->side ^ seg->color)) * dir;
     dcode->modifiers = MOD(ZBAR_MOD_GS1);
     return(ZBAR_DATABAR_EXP);
+abort:
+    return (ZBAR_NONE);
 }
 #undef IDX
 ```

bastien-roucaries, 2023-11-29:
@mchehab

risicle, 2023-11-29:
@bastien-roucaries I think exiting is a "better than nothing" strategy when it's not obvious what facilities a library has for "soft" error handling, but your patch does look better.

bastien-roucaries, 2023-12-01:
My patch was tested against POC. Return checksum error no overflow

jubalh, 2023-12-13:
> My patch was tested against POC.

Where can I find the POC?
I see it neither on this issue nor on https://hackmd.io/@cspl/B1ZkFZv23.

@bastien-roucaries do I see it right that in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1051724 you say that it is fixed for 0.23.90?
In https://security-tracker.debian.org/tracker/CVE-2023-40890 it looks like its only fixed in buster under version 0.22-1 (via patch) though.

bastien-roucaries, 2023-12-13:
I have asked privaltly to the reporter.

Yes it is fixed for debian unstable...0.23.90

jubalh, 2023-12-13:
> I have asked privaltly to the reporter.

I'll shoot you an email :)

> Yes it is fixed for debian unstable...0.23.90

But https://security-tracker.debian.org/tracker/CVE-2023-40890 sais its fixed for buster and unstable is still vulnerable?

carnil, 2023-12-13:
> > I have asked privaltly to the reporter.
> 
> I'll shoot you an email :)
> 
> > Yes it is fixed for debian unstable...0.23.90
> 
> But https://security-tracker.debian.org/tracker/CVE-2023-40890 sais its fixed for buster and unstable is still vulnerable?

The Debian security team did not had the poc so we prefer to mark it for now with the note as attached above. Once it's confirmed fixed upstream and this is aligned with the done change in unstable we can update the metadata.

jubalh, 2023-12-15:
The original reports:
* https://hackmd.io/@cspl/B1ZkFZv23
* https://hackmd.io/@cspl/H1PxPAUnn

have now been updated and include the POC.

carnil, 2023-12-16:
@jubalh should they be attached here in this issue separately as well in case the reference disapears and have it associated with this upstream issue clearly?

jubalh, 2023-12-18:
That might be a good idea. I considered attaching it as well but I didn't want to interfere with the reporters work. But I agree that it would be a better option!

mdevaev, 2024-01-07:
This is a disaster. A lot of software and high-level libraries use zbar, and this CVE is still not closed.

@mchehab Could you pay attention to this issue?

mdevaev, 2024-01-08:
@mchehab Sorry, I'm not ready to take responsibility for proposing and bringing to acceptance other people's patches for CVE for a codebase with which I am not familiar. It would be great if you did it yourself, or someone who has been following this problem here for a long time. I'm here solely as a user who is prevented by these CVEs from implementing QR support in my project.

jubalh, 2024-01-08:
@mchehab please see https://github.com/mchehab/zbar/pull/276

jubalh, 2024-01-08:
I'll attach the POCs which I got from @wqh17101 and which are now available on their website.

[image]
[image]

PR #276 (2024-01-09): Fix CVE-2023-40890 and CVE-2023-40889
To the best of my knowledge all the work was done by @bastien-roucaries / Remi Meier.
I just took the patches from https://salsa.debian.org/debian/zbar/-/tree/master/debian/patches and made them apply.

I hope this helps upstream to review the changes as requested in #263 (comment).

Fix #263
closed 2024-01-09 by mchehab