mchehab/zbar #277: Regression: Broken configure.ac in zbar 0.23.93 release (Qt5 related)

hosiet, 2024-01-09:
https://github.com/mchehab/zbar/blob/6ea7392beecac015f64ceea1ac11d33b83288e5f/configure.ac#L664-L683

The following issues (actually regressions) exist in this code block:

1. Hardcoded `pkg-config` reappears. We should use `$PKG_CONFIG` whenever possible, as seen in zbar <= 0.23.92.
1. The `AS_IF()` macro needs the `[` and `]` guarding for each of its parameter. In this code, the second parameter of `AS_IF()` has missing `[]` around the `PKG_CHECK_MODULES()` invocation.
1. The Qt5 detection logic is now broken, compared with zbar 0.23.92. The configure result would be like this:

```
checking for Qt5Core >= 5 Qt5Gui >= 5 Qt5Widgets >= 5.0 Qt5X11Extras >= 5.0... yes
Package Qt5 was not found in the pkg-config search path.
Perhaps you should add the directory containing `Qt5.pc'
to the PKG_CONFIG_PATH environment variable
Package 'Qt5', required by 'virtual:world', not found
configure: using Qt version 
configure: using moc from 
checking for javac... no
checking for jikes... no
```

This is due to Qt5 (at least in Linux distributions) never shipped a `Qt5.pc` anywhere. As a result, the invocation of `$PKG_CONFIG Qt5` will never succeed. Please revert to the old logic of detection `$MOC` and `$QT_VERSION` or seek other methods that correctly uses Qt5 pkg-config files.

jubalh, 2024-01-22:
It appears you have already patches ready at https://salsa.debian.org/debian/zbar/-/tree/master/debian/patches
Would you mind creating a PR?

mchehab, 2024-01-22:
> This is due to Qt5 (at least in Linux distributions) never shipped a `Qt5.pc` anywhere. As a result, the invocation of `$PKG_CONFIG Qt5` will never succeed. Please revert to the old logic of detection `$MOC` and `$QT_VERSION` or seek other methods that correctly uses Qt5 pkg-config files.

On Fedora, we have:
- a Qt5 pkg-config file is at /usr/lib64/pkgconfig/Qt5.pc (see qt5-qtbase-devel-5.15.11-7.fc39.x86_64);
- a Qt6 pkg-config file is at /usr/lib64/pkgconfig/Qt6.pc (see qt6-qtbase-devel-6.6.0-6.fc39.x86_64).

Perhaps the logic needs to be changed to try first Qt5.pc, then Qt.pc. If not found, fail back to the old way.

mchehab, 2024-01-22:
The upstream changes are there to better cope with installs were both Qt5 and Qt6 devel libs are installed. So, a simple solution like https://salsa.debian.org/debian/zbar/-/blob/master/debian/patches/0003-configure.ac-Use-old-way-to-detect-Qt5.patch may not be doing the right thing.

mchehab, 2024-01-22:
Ok, tested applying both patches from Debian: no breakages on Fedora. I'll apply them both.