mchehab/zbar #286: 0.23.93: problem with building zbar against Qt 6.x

kloczek, 2024-06-20:
I found in autoconf possibility to build zbar against Qt 6.x however looks like that possibility os not finished and it is even issue with correct Qt 6.x detection
```console
checking for /usr/bin/python3 extension module directory (pyexecdir)... ${PYTHON_EXEC_PREFIX}/lib64/python3.10/site-packages
checking for Python.h... yes
checking for gobject-introspection... yes
./configure: line 26913: QT_CFLAGS: command not found
./configure: line 26914: QT_LIBS: command not found
checking for Qt6... no
checking for Qt5Core >= 5 Qt5Gui >= 5 Qt5Widgets >= 5.0 Qt5X11Extras >= 5.0... no
checking for javac... /usr/bin/javac
```
I've started fiddlinfg with concigure.ac + I've added missing zbar-qt6.pc.in.
Here is my current patch
```patch
--- a/configure.ac
+++ b/configure.ac
@@ -664,12 +664,11 @@
 AS_IF([test "x$with_qt" != "xno"],
       [AS_IF([test "x$with_qt6" != "xno"],
             PKG_CHECK_MODULES([QT],
-                               [Qt6],
-                               [MOC=`pkg-config Qt6 --variable=moc`
-                                QT_VERSION=`$PKG_CONFIG Qt6 --modversion`
-                                QT6_HEADERS=`pkg-config Qt6 --variable=headerdir`
-                                CPPFLAGS="$CPPFLAGS -I$QT6_HEADERS -I$QT6_HEADERS/QtWidgets -I$QT6_HEADERS/QtCore -I$QT6_HEADERS/QtGui"
-                                qt_pkgconfig_file="zbar-qt5.pc"
+                               [Qt6Core Qt6Gui Qt6Widgets],
+                               [PKG_CHECK_VAR([MOC_PATH], [Qt6Core], [libexecdir])
+                                MOC=${MOC_PATH}/moc
+                                QT_VERSION=`$PKG_CONFIG Qt6Core --modversion`
+                                qt_pkgconfig_file="zbar-qt6.pc"
                                ],
                                [with_qt6="no"]))
        AS_IF([test "x$with_qt6" = "xno"],
--- a/zbar-qt6.pc.in
+++ b/zbar-qt6.pc.in
@@ -0,0 +1,12 @@
+prefix=@prefix@
+exec_prefix=@exec_prefix@
+libdir=@libdir@
+includedir=@includedir@
+
+Name: zbar-qt
+Description: bar code scanning and decoding Qt6 widget
+URL: http://zbar.sourceforge.net
+Version: @VERSION@
+Requires: zbar, Qt6Core >= 6, Qt6Gui >= 6
+Libs: -L${libdir} -lzbarqt
+Cflags: -I${includedir}
```
which still i not perfect because it allows detect Qt 6.x however it still prints some messy output
```console
checking for gobject-introspection... yes
./configure: line 26915: QT_CFLAGS: command not found
./configure: line 26916: QT_LIBS: command not found
checking for Qt6Core Qt6Gui Qt6Widgets... yes
./configure: line 26986: MOC_PATH: command not found
configure: using Qt version 6.7.1
configure: using moc from /usr/libexec/qt6/moc
```
Nevertheless with that patch build fails with:
```console
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/zbar-0.23.93'
/bin/sh ./libtool  --tag=CXX   --mode=compile /usr/bin/g++ -DHAVE_CONFIG_H -I. -I./include  -Iqt -I/usr/include/qt6/QtCore -I/usr/include/qt6 -DQT_CORE_LIB -I/usr/lib64/qt6/mkspecs/linux-g++ -I/usr/include/qt6/QtGui -DQT_GUI_LIB -I/usr/include/qt6/QtWidgets -DQT_WIDGETS_LIB -I./include -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -fPIC -Wall -Wno-parentheses -DLOCALEDIR=\"/usr/share/locale\" -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -mtls-dialect=gnu2 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -c -o qt/libzbarqt_la-QZBar.lo `test -f 'qt/QZBar.cpp' || echo './'`qt/QZBar.cpp
libtool: compile:  /usr/bin/g++ -DHAVE_CONFIG_H -I. -I./include -Iqt -I/usr/include/qt6/QtCore -I/usr/include/qt6 -DQT_CORE_LIB -I/usr/lib64/qt6/mkspecs/linux-g++ -I/usr/include/qt6/QtGui -DQT_GUI_LIB -I/usr/include/qt6/QtWidgets -DQT_WIDGETS_LIB -I./include -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -fPIC -Wall -Wno-parentheses -DLOCALEDIR=\"/usr/share/locale\" -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -mtls-dialect=gnu2 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -c qt/QZBar.cpp  -fPIC -DPIC -o qt/.libs/libzbarqt_la-QZBar.o
qt/QZBar.cpp:24:10: fatal error: QX11Info: No such file or directory
   24 | #include <QX11Info>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:1666: qt/libzbarqt_la-QZBar.lo] Error 1
```

joakim-tjernlund, 2025-04-09:
QT5 is going away.

joakim-tjernlund, 2025-04-09:
```
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/zbar-0.23.93'
/bin/sh ./libtool  --tag=CXX   --mode=compile /usr/bin/g++ -DHAVE_CONFIG_H -I. -I./include  -Iqt -I/usr/include/qt6/QtCore -I/usr/include/qt6 -DQT_CORE_LIB -I/usr/lib64/qt6/mkspecs/linux-g++ -I/usr/include/qt6/QtGui -DQT_GUI_LIB -I/usr/include/qt6/QtWidgets -DQT_WIDGETS_LIB -I./include -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -fPIC -Wall -Wno-parentheses -DLOCALEDIR=\"/usr/share/locale\" -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -mtls-dialect=gnu2 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -c -o qt/libzbarqt_la-QZBar.lo `test -f 'qt/QZBar.cpp' || echo './'`qt/QZBar.cpp
libtool: compile:  /usr/bin/g++ -DHAVE_CONFIG_H -I. -I./include -Iqt -I/usr/include/qt6/QtCore -I/usr/include/qt6 -DQT_CORE_LIB -I/usr/lib64/qt6/mkspecs/linux-g++ -I/usr/include/qt6/QtGui -DQT_GUI_LIB -I/usr/include/qt6/QtWidgets -DQT_WIDGETS_LIB -I./include -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -fPIC -Wall -Wno-parentheses -DLOCALEDIR=\"/usr/share/locale\" -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -mtls-dialect=gnu2 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -c qt/QZBar.cpp  -fPIC -DPIC -o qt/.libs/libzbarqt_la-QZBar.o
qt/QZBar.cpp:24:10: fatal error: QX11Info: No such file or directory
   24 | #include <QX11Info>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:1666: qt/libzbarqt_la-QZBar.lo] Error 1
```
There is an qtx11extras_p.h in QtGui that looks similar

callegar, 2025-05-05:
Current zbarcam-qt has issues and dumps core when run as a wayland application.