mchehab/zbar #288: Potential bug in _zbar_image_copy function ?

khoaphamce, 2024-07-23:
Hi authors,

I found the potential bug from _zbar_image_copy function in zbar/image.h (line 158 -> 172):
```C
int i, len = src->datalen;
long *sp = (void *)src->data, *dp = (void *)dst->data;
char *spc, *dpc;

/* Do it word per word, in order to speedup */
for (i = 0; i < len; i += sizeof(long))
	  *dp++ = ~(*sp++);

// (1): i is now larger or equal to len (i >= len)

/* Deal with non-aligned remains, if any */
len -= i; // (2): from (1) => (len <= 0)
spc = (char *)sp;
dpc = (char *)dp;
for (i = 0; i < len; i++) //(3): from (2) => this loop will never be executed
	  *dpc++ = ~(*spc++);
```
The for loop to `/* Deal with non-aligned remains, if any */` will never be executed since the loop above `/* Do it word per word, in order to speedup */` make an additional of `i += sizeof(long)` right before exit the loop, this will result in `i >= len` in every situation. The loop to `/* Deal with non-aligned remains, if any */` will never run because of the described behaviour.

May I create a PR to fix this ?

Thank you.

jasp00, 2024-08-29:
Current code does not deal with non-aligned remains correctly, so you may prepare a pull request.

khoaphamce, 2024-08-30:
> Current code does not deal with non-aligned remains correctly, so you may prepare a pull request.

I created the PR: https://github.com/mchehab/zbar/pull/295 , please help check it out.
Thanks.