mchehab/zbar #292: Feature: scan image from clipboard or screen

joshgold22, 2024-08-23:
Would it be possible (or is it already) to read an image directly from the clipboard without having to save as a file (might use a temp file internally)

Similarly, would it be possible to just run a zbar command when there's a QR visible somewhere on the screen, without even copying to clipboard?

danielwerg, 2024-08-26:
If you are use wl-clipboard you can do this:

```sh
wl-paste | zbarimg -
```

and

```sh
slurp | grim -g - - | zbarimg - --raw -q
```

joshgold22, 2024-08-26:
Thanks for the idea, I think this will work analogously as a workaround on mac osx — I was able to use it to dump the raw png file data to a pipe.
getclipb PNGf | ...
https://apple.stackexchange.com/a/375353/19289
macos - What's the quickest way to get a graphic on the clipboard saved to disk? - Ask Different

I can't confirm, though, because I'm new to zbar, and I see zbarimg is not working for me even with image files saved on disk, (including -S\*.enable)

(Assuming I get zbar woriking, I still think it could be nice to have these two capacities built in, but perhaps they would require a lot of OS-specific tweaking.)