mchehab/zbar #309: zbarcam-qt dumps core

callegar, 2025-05-04:
I see it on manjaro (arch), with version 0.23.93.

It is sufficient to start the application to have it dump core.

```
> zbarcam-qt
Segmentation fault (core dumped)
```
The gtk client works.

callegar, 2025-05-04:
This seems to be well know. Apparently some linux distro disable building qt altogether (https://lists.nongnu.org/archive/html/guix-commits/2025-01/msg05598.html).

callegar, 2025-05-05:
Important note: when run as a wayland application. In X11 mode it works.

callegar, 2025-05-09:
I think that you need the equivalent of Commit 86e15ee for zbarcam-qt.

MorsMortium, 2025-07-24:
QT_QPA_PLATFORM=xcb zbarcam-qt works on wayland

MorsMortium, 2025-07-24:
True, it can also set the env var for itself, I've seen that once
`char env[] = "QT_QPA_PLATFORM=xcb";
    putenv(env);
`
Was just writing for those that couldn't find how to force the app, until it gets fixed.

mchehab, 2026-10-01:
I added a patch fixing it for Wayland:
https://github.com/mchehab/zbar/commit/85a3b6d73a1e58fb597ea044721b1d0090bf86b4

So, I'm closing this one. Please reopen it if this doens't help - together with the exact environment (x11, wayland?) and a gdb dump.

closed 2026-10-01 by mchehab