mchehab/zbar #313: How can zbarcam decode a QR code that zbarimg can't?

Sadi58, 2025-06-28:
I have a shell script that uses the command `zbarimg -q --raw "$filename" 2>/dev/null` to convert a QR image to text, but it fails with some WiFi QR codes generated by my Samsung Galaxy phone.
However, when I tried `zbarcam`, launching it from terminal, and showing the image to my laptop camera, terminal output displayed the text content of the QR image!
I guess I would get a similar result, if not better, if I used the Open button in GUI to get the QR image file decoded, but I was looking for a fallback method in my shell script.   
Unfortunately, I couldn't use the image file as input for `zbarcam` from the command line.
So, I ended up using Python qreader instead, with tons of dependencies.
It would be so useful if we could make use of those processes in `zbarcam` to decode QR images input from the command line as well.
Maybe I just couldn't use the right key to open that door?

mchehab, 2026-10-01:
that sounds weird, as both `zbarcam` and `zbarimg` uses exactly the same library to decode the image. There was a bug on QR decoding that could be the root cause (It was doing a wrong swap on some cases). Perhaps it is related to it.

Sadi58, 2026-10-01:
I looked into this again.
I encountered the problem with one such WiFi QR code when I used the command `zbarimg -q --raw "$filename" 2>/dev/null` although it worked with another one.
However, I was able to overcome the problem by using the `command zbarimg -q --raw <(convert "$filename" -colorspace Gray -normalize -contrast-stretch 2%x1% -threshold 50% - 2>/dev/null)`.
I suspect the conversion in the second command fixes an odd problem which might perhaps be related with 63-character WiFi password.

closed 2026-10-01 by Sadi58